RADARC/Scout-Radio #4: Downloading of patch.csg is slow

johngumb, 2025-03-12:
Improve downloading of patch.csg. Ideas included speeding up I2C and embedding patch in python rather than having a separate file. Perhaps a pre-compiled .mpy file may help.

ribbotson, 2025-03-17:
Running the I2C at maximum speed400kHz to the Si4732 does not appear to make significant improvement.

johngumb, 2025-03-17:
great ideas, thanks Richard. We have some easy things to try. Considering breaking out the logic analyser to see where the long silences are :-) Need to clear the bench first...that's a bigger problem.

johngumb, 2025-03-20:
better as of c342cbc2423 (ish)
patch.csg convered to a pseudo-compressed binary form using the usual "compression" method for these things where the first byte of an eight byte sequence is either 0x15 or 0x16.

johngumb, 2025-03-26:
claiming fixed.

closed 2025-03-26 by johngumb