RADARC/Scout-Radio #18: Consider standardising on pico W

johngumb, 2025-03-17:
Consider standardising on pico W.

More fun for scouts and us.

ribbotson, 2025-03-17:
I have Pico W, Pico 2 and Pico 2 W with headers if you want to try. Interested to see if the 2.4GHz interferes with the radio.
There were some concerns about an internet or other network (meshtastic) connected device for young people. The other Picos should just be plug in replacements.

johngumb, 2025-03-17:
Thanks Richard. I suppose we could lock down internet connectivity. There again no doubt there are clever scouts. Got some reading up to do on exactly what the W variant can do (bluetooth, BLE etc.).

johngumb, 2025-03-17:
I'd (almost) be prepared to lock down to bluetooth/BLE if necessary to get W variant.